mr-manuel/venus-os_dbus-mqtt-grid #2: Feature-Idea

kupfermuetze, 2023-04-23:
Currently I'm using to connect my tasmota smart-meter into the victron universe. As far as I understodd (at least this is how i do it currently) I run an additional service which transferes the topic from tasmota to the desired format.

It would be perfect if one could state each single topic in the config file; this would increase flexibility and reduces complexity (in the application).

mr-manuel, 2023-04-23:
Hi,

what format does your Tasmota publish to MQTT?

Is it publishing all values to one topic or to multiple topics? Multiple topics is not ideal.

kupfermuetze, 2023-04-23:
Hello (wow, that was a quick response),
 
I get a full set every 10s :
{"Time":"2023-04-23T19:19:32","haushalt�":{"bezug":17448.698566,"einspeisung":9594.207150,"leistung_gesamt":103.810000,"leistung_L1":276.800000,"leistung_L2":-222.440000,"leistung_L3":46.470000}}

but to have a smooth operation i would like to have the values that are transmitted each second (approx.) 
{"Time":"2023-04-23T19:19:32","haushalt�":{"leistung_L3":46.490000}}
{"Time":"2023-04-23T19:19:32","haushalt�":{"leistung_L2":-219.460000}}
{"Time":"2023-04-23T19:19:32","haushalt�":{"leistung_L1":272.820000}}
{"Time":"2023-04-23T19:19:32","haushalt�":{"leistung_gesamt":94.460000}}
{"Time":"2023-04-23T19:19:32","haushalt�":{"einspeisung":9594.207150}}
{"Time":"2023-04-23T19:19:32","haushalt�":{"bezug":17448.698566}}

mr-manuel, 2023-04-23:
The values that are send every 10 seconds would be better, but as you said, it's to slow for a smooth operation.

Are the values that are send every second send to the same topic, so that the previous value is overridden?

kupfermuetze, 2023-04-23:
Yes, Ther topic "Haushalt" in my case ist overwritten with each new value

mr-manuel, 2023-04-23:
Ok. I will not implement this right now, since it would require a lot of work. Are you using Node-RED on the Venus OS Large to translate the values or how you do it?

If a lot of people are requesting it, I will think about it again. Please give the first comment a thumbs up for this.

kupfermuetze, 2023-04-23:
OK, understood. 

Currently I'm using my FHEM instance for the Translation. Your script runs in a Cerbo GX.

mr-manuel, 2023-04-23:
If you want you can fork my repo and open a PR to add this format.

mr-manuel, 2023-05-22:
Currently not planned.

randomname32, 2023-10-10:
@kupfermuetze I created a PR for this issue.
I am configured tasmota to publish directly to the mqtt server on venusOS

mr-manuel, 2023-10-24:
The PR was merged and the `README.md` was updated.

closed 2023-10-24 by mr-manuel