mr-manuel/venus-os_dbus-mqtt-grid #4: Feature-Request: Check healthyness

gvzdus, 2023-05-23:
In case the feed of grid power information stops (e.g.: Wifi trouble, whatever), the ESS would either increase battery charging / discharging to the maximum, unless the seldom case would happen that the last grid power was "perfect".
Thinking about the best solution, the monitoring of the feed should happen on the venus/GX device itself. IMHO, the best would be a timer, a boolean and 2 commands:
- "after_x_seconds_no_data_feed_set_cmd_1" (e.g.: Turn ESS off)
- "when_feed_is_present_again_set_cmd_2" (e.g., Turn ESS on)
The alternative would be an external monitoring script on the GX/venus.

mr-manuel, 2023-05-23:
You cannot turn on/off the ESS as far as I know. Maybe you meant the MultiPlus or Quattro?

This can easily archived with Node-RED that is in the Venus OS Large image. Here an example that turns off the MultiPlus, if not receiving data for 60 seconds. But I would recommend to fix the connectivity problems.

[image]

```json
[
    {
        "id": "94e680ff8884460b",
        "type": "victron-input-gridmeter",
        "z": "202a089e4ad5eef4",
        "service": "com.victronenergy.grid/31",
        "path": "/Ac/L1/Power",
        "serviceObj": {
            "service": "com.victronenergy.grid/31",
            "name": "MQTT Grid"
        },
        "pathObj": {
            "path": "/Ac/L1/Power",
            "type": "float",
            "name": "L1 Power (W)"
        },
        "name": "",
        "onlyChanges": false,
        "x": 150,
        "y": 860,
        "wires": [
            [
                "72a48ef4f99d636e"
            ]
        ]
    },
    {
        "id": "72a48ef4f99d636e",
        "type": "trigger",
        "z": "202a089e4ad5eef4",
        "name": "",
        "op1": "3",
        "op2": "4",
        "op1type": "num",
        "op2type": "num",
        "duration": "60",
        "extend": true,
        "overrideDelay": false,
        "units": "s",
        "reset": "",
        "bytopic": "all",
        "topic": "topic",
        "outputs": 1,
        "x": 390,
        "y": 860,
        "wires": [
            [
                "f680444645e78ba9"
            ]
        ]
    },
    {
        "id": "f680444645e78ba9",
        "type": "victron-output-vebus",
        "z": "202a089e4ad5eef4",
        "service": "com.victronenergy.vebus/275",
        "path": "/Mode",
        "serviceObj": {
            "service": "com.victronenergy.vebus/275",
            "name": "MultiPlus-II 48/5000/70-50"
        },
        "pathObj": {
            "path": "/Mode",
            "type": "enum",
            "name": "Switch Position",
            "remarks": "<p>Note that <tt>/ModeIsAdjustable</tt> needs to be set to 1.</p> ",
            "enum": {
                "1": "Charger Only",
                "2": "Inverter Only",
                "3": "On",
                "4": "Off"
            },
            "writable": true
        },
        "initial": "",
        "name": "",
        "onlyChanges": false,
        "x": 720,
        "y": 860,
        "wires": []
    }
]
```

gvzdus, 2023-05-23:
Yes, I have a MultiPlus II and a (real) Cerbo GX. What I would like to do is exactly what you propose: Setting Mode to 4 ("off") in case of "feed failure". Yes, a solution with an external SW component like NodeRed (me running FHEM historically) is possible, but the most bullet-proof solution would be IMHO directly on the Cerbo GX.

mr-manuel, 2023-05-23:
This solution is bulletproof. With Venus OS Large you have Node-RED directly on the Cerbo GX.

gvzdus, 2023-05-23:
Got it, thank you. I did not know about Venus OS Large.

closed 2023-05-23 by mr-manuel