mr-manuel/venus-os_dbus-mqtt-grid #6: FeatureRequest: PV meter

Jambola77, 2023-08-02:
Hello,
I habe an old PV Inverter which is not capable to communicate with the Cerbo. It would be great to make it possible to add a possibility to send the mqqt values from the inverter to the Cerbo
JSON is: /tele/WR1/PAC: 10.0 (e.g.)

mr-manuel, 2023-08-02:
Did I understand right that the PV inverter publishes the value `10.0` with the topic `/tele/WR1/PAC`?

You can use Node-RED that is contained in the Venus-OS large to fetch the data and send it to the right topic you need.

Jambola77, 2023-08-02:
Hello manuel,
yes, topic is /tele/WR1/PAC.
I use a Cerbo-S GX, what do you mean with Node-RED?

mr-manuel, 2023-08-02:
Just Google it.

https://www.victronenergy.com/live/venus-os:large

closed 2023-09-12 by mr-manuel