mr-manuel/venus-os_dbus-mqtt-grid #7: Was passiert bei einem Verbindungsverlust

BenjaminK87, 2023-09-02:
Wenn ich es richtig sehe, wird das Skript beendet.
Was macht denn ein Victron System wenn man die RS485 zwischen Zähler und dem GX Gerät trennt?
Wäre es nicht sinnvoll Laden/Entladen vom Gerät zu deaktivieren? 

Wenn gerade viel Strom aus der Batterie gezogen wird und dann die Verbindung abbricht, dann zieht der Multiplus schön den ganzen Akku leer, egal wie sich der Verbrauch weiter geändert hat.

mr-manuel, 2023-09-02:
Meines Wissens nach läuft es ganz normal weiter, da das Grid Meter kein essentieller Bestandteil des Systems für den Betrieb ist. Du kannst dir mit Node-RED selbst was bauen, dass das Laden/Entladen der Batterie stoppt, wenn das Grid Meter nicht mehr erreichbar ist.

Mit der Frage bist du jedoch besser in der Victron Community aufgehoben. Poste am Besten den Link zu deiner Frage hier, dann können es andere auch nachvollziehen.

closed 2023-09-12 by mr-manuel

randomname32, 2023-10-10:
@BenjaminK87 Hast du eine Lösung finden können?

BenjaminK87, 2023-10-11:
@randomname32 Nee ich hatte nicht mehr weiter nachgeforscht. Läuft bis jetzt auch so. Nur ärgerlich wenn die Kommunikation dann doch mal abbricht und der Akku mit Strom aus dem Netz aufgeladen wird.

fischan, 2023-10-11:
@randomname32 und @BenjaminK87 
ich kann ja mal meine Lösung posten, ich denke das der Input von @mr-manuel stammt. Ich weiß leider nur nicht mehr wo ich das gefunden habe.

Im Prinzip brauchst du Venus OS Large mit aktivierten Node Red.
Und dann machst du einen Flow mit mindestens 3 Nodes.
Den Input liefert dabei das Grid Meter (Zahlerschrank in meinem Beispiel) dieser sendet neue Leistungswert an einen Trigger weiter.
Der Trigger ist so konfiguriert das er den Wert "3" Sendet wenn mindestens alle 15 Sekunden ein Wert vom Trigger empfangen wird. Sollte hier keine Wert für einen Zeitraum von 15 Sekunden ankommen, dann sendet der Trigger den Wert 4.
Der Empfänger der 3 bzw. 4 ist mein Multiplus II.
3 = Schalterposition ON
4 = Schalterpostition OFF

Das ist im Prinzip die einfachste Version.
Sinnvoll wäre es z.B. auch noch eine Email oder Telegram Nachricht zu verschicken wenn durch den Flow z.B. der Multiplus ausgeschaltet wurde.

Hier der Export der 3 Nodes
[{"id":"94e680ff8884460b","type":"victron-input-gridmeter","z":"1e11c8b7a5bac96d","service":"com.victronenergy.grid/31","path":"/Ac/L1/Power","serviceObj":{"service":"com.victronenergy.grid/31","name":"Zaehlerschrank"},"pathObj":{"path":"/Ac/L1/Power","type":"float","name":"L1 Power (W)"},"name":"","onlyChanges":false,"x":150,"y":40,"wires":[["72a48ef4f99d636e"]]},{"id":"72a48ef4f99d636e","type":"trigger","z":"1e11c8b7a5bac96d","name":"","op1":"3","op2":"4","op1type":"num","op2type":"num","duration":"15","extend":true,"overrideDelay":false,"units":"s","reset":"","bytopic":"all","topic":"topic","outputs":2,"x":370,"y":40,"wires":[["f680444645e78ba9"],["f680444645e78ba9","1aaaa33af117ec7d"]]},{"id":"f680444645e78ba9","type":"victron-output-vebus","z":"1e11c8b7a5bac96d","service":"com.victronenergy.vebus/276","path":"/Mode","serviceObj":{"service":"com.victronenergy.vebus/276","name":"MultiPlus-II 48/5000/70-50"},"pathObj":{"path":"/Mode","type":"enum","name":"Switch Position","remarks":"<p>Note that <tt>/ModeIsAdjustable</tt> needs to be set to 1.</p> ","enum":{"1":"Charger Only","2":"Inverter Only","3":"On","4":"Off"},"writable":true},"initial":"","name":"","onlyChanges":true,"x":670,"y":40,"wires":[]}]

randomname32, 2023-10-12:
Vielen Dank für die schnellen Antworten!

Das mit dem Node-Red script sieht ja easy aus. Das müsste doch auch mit dem python script gehen, oder? Ich hab mal angefangen was zu schreiben und erstell dann einen PR wenns fertig ist.

https://github.com/randomname32/venus-os_dbus-mqtt-grid/blob/feature/disable-mp2-on-timeout/dbus-mqtt-grid/dbus-mqtt-grid.py#L528

fischan, 2023-10-15:
@randomname32 
gerne, ich kann leider kein python, aber sollte vermutlich schon klappen. Aber ich werde bei der Node Red Lösung im Venus OS bleiben, ich habe da mittlerweile einige Flows und auch eine Visu. Finde das ganz gut :-)

randomname32, 2024-01-02:
Ich habe jetzt auch die NodeRed Lösung verwendet. Ist doch deutlich einfacher und flexibler als die Python Lösung ;-)

Allerdings hat die Lösung mit dem setzen von "4: Off" für mich einen Nachteil: der Notstrom wird abgestellt. Und daran hängt mein VenusOS-Raspi und der Gefrierschrank...
Ich habe den Flow deswegen so geändert, dass statt "4: Off" der Multiplus auf "1: Charger Only" gestellt wird.
Außerdem setze ich parallel "Active Input Current Limit (A)" auf den Wert "0".
Wird die Verbindung wieder hergestellt setze ich das input current limit wieder zurück auf 50A (mein default Wert)