mr-manuel/venus-os_dbus-mqtt-grid #10: Home Assistant example

Handy-Harry, 2023-11-15:
I have an usb dsmr meter in my rpi with  Home Assistant. Could you give me an example how to mqtt the value's to Venusos?

mr-manuel, 2023-11-16:
Unfortunately I‘m not using Home Assistant. Maybe someone other can help you with that.

Steamerzone, 2023-11-26:
I'm using a vue2 energy meter flashed to esphome as a energymeter, I use the following automation in HA to push the values to mqtt:

`alias: mqtt publish sensor value grid power
description: ""
trigger:
  - platform: state
    entity_id: sensor.vue2_total_overall_power
condition: []
action:
  - service: mqtt.publish
    data_template:
      payload: |
        {
          "grid": {
            "power": {{ (states('sensor.vue2_total_overall_power') | float(0)) - (states('sensor.vue2_total_overall_power_return') | float(0)) }},
            "L1": {
              "power": {{ (states('sensor.vue2_phase_a_power') | float(0)) - (states('sensor.vue2_phase_a_power_return') | float(0)) }},
              "voltage": {{ states('sensor.vue2_phase_a_voltage') }}
            },
            "L2": {
              "power": {{ (states('sensor.vue2_phase_b_power') | float(0)) - (states('sensor.vue2_phase_b_power_return') | float(0)) }},
              "voltage": {{ states('sensor.vue2_phase_b_voltage') }}
            },
            "L3": {
              "power": {{ (states('sensor.vue2_phase_c_power') | float(0)) - (states('sensor.vue2_phase_c_power_return') | float(0)) }},
              "voltage": {{ states('sensor.vue2_phase_c_voltage') }}
            }
          }
        }
      topic: homeassistant/energy/grid`

mr-manuel, 2023-12-01:
@Steamerzone thanks for helping with this issue :-)

closed 2024-01-29 by mr-manuel

jezzaf, 2024-03-13:
Hi, I'm hoping someone can give me some help with this. 
I have data going into Venus but i get an Error. Exception occurred: float division by zero

I'm wondering if i can get an example of the config for just having 1 single meter on mains (single phase) with voltage. 

I can then look at adding the circuits i have later. 

thanks

mr-manuel, 2024-03-13:
Please open a new issue and share the whole error log

7-er, 2025-03-13:
Hello Mr. Manuel, 
please tell me, where I can change the topic names "grid" and "power" to my own ones. 

With Thanks and nice regards from northern Germany
Siemer