mr-manuel/venus-os_dbus-mqtt-grid #13: Some more Info for direct connection Tasmota->Venus

BenjaminK87, 2023-12-17:
It took me a while to get it running without a broker in between because of the following things:

- The Tasmota Script need to have "grid" Name for the sensor in the +1,3s,0,9600,grid line. Otherwise the script won't recognize the messages.
```
>D
>B
=>sensor53 r
>M 1
+1,3,s,0,9600,grid
1,77070100010800ff@1000,Bezug,kWh,energy_forward,19
1,77070100100700ff@1,Aktueller Verbrauch,W,power,16
1,77070100020800ff@1000,Einspeisung,kWh,energy_reverse,19
#
```

- The topic in tasmota has to fit together with the config.ini. In my case i chose:
Tasmota Topic: SML
Tasmota Full Topic: %topic%/

config.ini topic = SML/SENSOR

And another side note: for my installation the VRM looked strange with only transferring grid_power and not the single phases. I changed the script, so that if only grid_power is received, all the phases are calculated in the script.

After the "check if L3 and L3 -> power exists i added the following:
```
if grid_L1_power is None:
                            grid_L1_power = round(grid_power/3.0,2)
                            grid_L1_voltage = grid_voltage
                            grid_L1_current = round(grid_power/3.0/grid_L1_voltage,2)
                            grid_L2_power = round(grid_power/3.0,2)
                            grid_L2_voltage = grid_voltage
                            grid_L2_current = round(grid_power/3.0/grid_L2_voltage,2)
                            grid_L3_power = round(grid_power/3.0,2)
                            grid_L3_voltage = grid_voltage
                            grid_L3_current = round(grid_power/3.0/grid_L3_voltage,2)
```

I also commented the "Received JSON MQTT message...." because i receive the energy_forward, energy_reverse not at the same time as the power value.

Because of this it was also necessary to change the setting dbusservice

```
if grid_forward is not None: 
                self._dbusservice["/Ac/Energy/Forward"] = (
                    round(grid_forward, 2)
                )
            if grid_reverse is not None: 
                self._dbusservice["/Ac/Energy/Reverse"] = (
                    round(grid_reverse, 2)
                )
```

Maybe the infos are useful for someone.
Thank you for the great script!

mr-manuel, 2023-12-21:
Did I understand correctly that you have a three phase system, but provided only the sum of power pf all three phases and it looked strange? If you provide only the total power it will be used for L1. If you have a three phase system you have to provide all three phases. If you calculate this values for each phase you will mess up your system, since the values are not real.

BenjaminK87, 2023-12-22:
I only have one MP2. So 1 phase system. 
I didn't make some screenshots but the AC grid showed strange values for L2 and L3.

closed 2024-01-29 by mr-manuel