mr-manuel/venus-os_dbus-mqtt-grid #14: Driver sometimes stuck in initial loop.

gdave321, 2023-12-18:
I've seen couple of times that the drivers seems to get stuck in initial loop setup, during venus-OS upgrade/reboot this sometimes happens and also when eg. the (external) mqtt broker has been offline for sometime due to eg. maintenance.

Log shows this:
@40000000657f677f27e831b4 *** starting dbus-mqtt-grid ***
@40000000657f67be1efd307c WARNING:root:Waiting since 60 seconds for receiving fi
rst data...
@40000000657f67fa21ec6ce4 WARNING:root:Waiting since 120 seconds for receiving first data...
@40000000657f683625390b8c WARNING:root:Waiting since 180 seconds for receiving first data...
@40000000657f687228a15824 WARNING:root:Waiting since 240 seconds for receiving first data...
and this can go on for extended time.

So I added few lines of code which seems to work rather well, probably this timeout should re-use the timeout in ini file for as a quick hack it does it job just fine. Would be nice if you could add this or something similar to this driver (BTW same applies to dbus-mqtt-pv).

```--- dbus-mqtt-grid.py.ORIG
+++ dbus-mqtt-grid.py
@@ -570,6 +570,11 @@
             )
         sleep(5)
         i += 1
+        if i > 120:
+            logging.error(
+                "Driver stopped. waiting too long for first data (>600sec)"
+            )
+            sys.exit()
 
     # formatting
     def _kwh(p, v):
```

Now log looks like:
@40000000657f699e3693b3dc WARNING:root:Waiting since 540 seconds for receiving first data...
@40000000657f69da3a09141c WARNING:root:Waiting since 600 seconds for receiving first data...
@40000000657f69df3a5ff64c ERROR:root:Driver stopped. waiting too long for first data (>600sec)
@40000000657f69e007bb06dc *** starting dbus-mqtt-grid ***

gdave321, 2023-12-18:
Yes, first time I found this it was several hours. At the time I just killed the driver and that immediately got it to work again.  Seems the mqtt data will never arrive anymore, so it very well could be that the mqtt library is to blame?

mr-manuel, 2023-12-18:
Ok, I will add the timeout here. Thanks you for reporting!

mr-manuel, 2023-12-18:
A new version with a fix was released.