mr-manuel/venus-os_dbus-mqtt-grid #18: Homeassistant help and error

jezzaf, 2024-03-13:
Hi, I'm hoping someone can give me some help with this.
I have data going into Venus but i get an Error. Exception occurred: float division by zero

2024-03-13 09:31:54.565327500 ERROR:root:Exception occurred: float division by zero

I'm wondering if i can get an example of the config for just having 1 single meter on mains (single phase) with voltage.

thanks

mr-manuel, 2024-03-13:
Is this all you get from the log? Please enable debugging in the config file and post the whole log.

jezzaf, 2024-03-13:
I'm not sure what i did but now that error is gone and I'm now getting this.

2024-03-13 20:38:16.244660500 INFO:root:Waiting 5 seconds for receiving first data...
2024-03-13 20:38:19.789378500 ERROR:root:Received message is not a valid JSON. Expecting property name enclosed in double quotes: line 7 column 5 (char 94)
2024-03-13 20:38:19.790156500 DEBUG:root:MQTT payload: '{\n  "grid": {\n    "power": 734.4,\n    "voltage": 235.0,\n    "L1": {\n      "power": 734.4,\n    }\n  }\n}'

in homeassistant i have this 

      payload: |
        {
          "grid": {
            "power": {{ (states('sensor.total_power') | float(0)) }},
            "voltage": {{ (states('sensor.house_volts') | float(0)) }},
            "L1": {
              "power": {{ (states('sensor.power_total_used') | float(0)) }},
            }
          }
        }
      topic: homeassistant/energy/grid

it also looks like the wattage is only populating the AC Loads and not the incoming from the grid.

[image]

Thanks

mr-manuel, 2024-03-13:
This is not a valid JSON, after each `,` has to be some data.

```
    {
      "grid": {
        "power": {{ (states('sensor.total_power') | float(0)) }},
        "voltage": {{ (states('sensor.house_volts') | float(0)) }},
        "L1": {
          "power": {{ (states('sensor.power_total_used') | float(0)) }}, <-- remove this comma ","
        }
      }
    }
```

jezzaf, 2024-03-13:
thankyou.

i now have this 
2024-03-13 22:33:30.132169500 DEBUG:root:Grid: 457.2 W - 0.0 V - 0.0 A
2024-03-13 22:33:34.425827500 ERROR:root:Exception occurred: 'voltage'
2024-03-13 22:33:34.425977500 DEBUG:root:MQTT payload: '{\n  "grid": {\n    "power": 446.2,\n    "voltage": 240.7,\n    "L1": {\n      "power": 785.8\n    }\n  }\n}'

jezzaf, 2024-03-16:
I'm not sure what is going on. it looks like the voltage is in the payload but it doesn't seam to be able to get that data. 

2024-03-16 02:05:35.106367500 DEBUG:root:MQTT payload: '{\n  "grid": {\n    "power": -3954.8,\n    "voltage": 239.1,\n    "energy_forward": 2042.05,\n    "energy_reverse": 1881.15\n  }\n}'
2024-03-16 02:05:35.392000500 DEBUG:root:Grid: -3954.8 W - 0.0 V - 0.0 A
2024-03-16 02:05:40.099858500 ERROR:root:Exception occurred: 'voltage'

it also doesn't seam to populate the AC input correctly.
[image]

any thoughts? 

thanks

mdkeil, 2024-03-16:
I would try to set the default value for the voltage to 230 or 240.

    ```
{
      "grid": {
        "power": {{ (states('sensor.total_power') | float(0)) }},
        "voltage": {{ (states('sensor.house_volts') | float(240)) }},
        "L1": {
          "power": {{ (states('sensor.power_total_used') | float(0)) }}
        }
      }
    }
```

jezzaf, 2024-03-16:
thanks but unfortunately its still the same 

2024-03-16 10:29:15.208258500 ERROR:root:Received JSON MQTT message does not include a grid object. Expected at least: {"grid": {"power": 0.0}"}
2024-03-16 10:29:15.208427500 DEBUG:root:MQTT payload: '{\n  "Grid": {\n    "power": 4394.8,\n    "voltage": 236.4,\n    "energy_forward": 2048.08,\n    "energy_reverse": 1893.34\n  }\n}'
2024-03-16 10:29:15.789926500 DEBUG:root:Grid: 1101.8 W - 0.0 V - 0.0 A

this is what I'm sending 
      payload: |
        {
          "Grid": {
            "power": {{ (states('sensor.total_power') | float(0)) }},
            "voltage": {{ (states('sensor.house_volts') | float(235)) }},
            "energy_forward": {{ (states('sensor.total_import_kwh') | float(0)) }},
            "energy_reverse": {{ (states('sensor.total_export_kwh') | float(0)) }}
          }
        }

mr-manuel, 2024-03-16:
Now you wrote `Grid` instead of `grid`.

jezzaf, 2024-03-16:
sorry so many copy and pasts trying different combinations 

      payload: |
        {
          "grid": {
            "power": {{ (states('sensor.total_power') | float(0)) }},
            "voltage": {{ (states('sensor.house_volts') | float(0)) }},
            "energy_forward": {{ (states('sensor.total_import_kwh') | float(0)) }},
            "energy_reverse": {{ (states('sensor.total_export_kwh') | float(0)) }}
          }
        }

2024-03-16 12:00:45.229360500 ERROR:root:Exception occurred: 'voltage'
2024-03-16 12:00:45.229911500 DEBUG:root:MQTT payload: '{\n  "grid": {\n    "power": 1314.0,\n    "voltage": 239.0,\n    "energy_forward": 2050.48,\n    "energy_reverse": 1893.34\n  }\n}'
2024-03-16 12:00:45.359971500 DEBUG:root:Grid: 1314.0 W - 0.0 V - 0.0 A

mr-manuel, 2024-03-16:
Could you download the script from the repository again? I added some error handling, so we get more error informations.

mdkeil, 2024-03-16:
Did you tryed it with float(240) for the voltage-value?

jezzaf, 2024-03-16:
Hi 
yeah i did try with the 240 but it didn't change anything. 

i now have a bunch of disconnect warnings along with the same as before. 

2024-03-16 23:23:28.083379500 INFO:root:MQTT client: Connected to MQTT broker!
2024-03-16 23:23:30.176309500 ERROR:root:Exception occurred: 'voltage'
2024-03-16 23:23:30.176855500 DEBUG:root:MQTT payload: '{\n  "grid": {\n    "power": 535.8,\n    "voltage": 236.9,\n    "energy_forward": 2057.97,\n    "energy_reverse": 1893.34\n  }\n}'
2024-03-16 23:23:30.547810500 WARNING:root:MQTT client: Got disconnected
2024-03-16 23:23:30.548281500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-03-16 23:23:30.555218500 DEBUG:root:Grid: 535.8 W - 0.0 V - 0.0 A
2024-03-16 23:23:31.564441500 INFO:root:MQTT client: Connected to MQTT broker!
2024-03-16 23:23:32.585260500 WARNING:root:MQTT client: Got disconnected
2024-03-16 23:23:32.585720500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-03-16 23:23:33.605601500 INFO:root:MQTT client: Connected to MQTT broker!
2024-03-16 23:23:34.625633500 WARNING:root:MQTT client: Got disconnected

jezzaf, 2024-03-16:
now the disconnection ones have gone and left with the same as before 

2024-03-16 23:40:00.179783500 ERROR:root:Exception occurred: 'voltage'
2024-03-16 23:40:00.179789500 DEBUG:root:MQTT payload: '{\n  "grid": {\n    "power": 1109.6,\n    "voltage": 236.5,\n    "energy_forward": 2058.18,\n    "energy_reverse": 1893.34\n  }\n}'
2024-03-16 23:40:01.133523500 DEBUG:root:Grid: 1109.6 W - 0.0 V - 0.0 A

mdkeil, 2024-03-17:
Can you please post your entire hass-part of the mqtt-publish?

jezzaf, 2024-03-17:
Sure 

alias: Mains to MQTT
  description: ''
  trigger:
  - platform: state
    entity_id:
    - sensor.total_power
  condition: []
  action:
  - service: mqtt.publish
    data_template:
      payload: |
        {
          "grid": {
            "power": {{ (states('sensor.total_power') | float(0)) }},
            "voltage": {{ (states('sensor.house_volts') | float(0)) }},
            "energy_forward": {{ (states('sensor.total_import_kwh') | float(0)) }},
            "energy_reverse": {{ (states('sensor.total_export_kwh') | float(0)) }}
          }
        }
      topic: homeassistant/energy/grid

jezzaf, 2024-03-17:
alias: Mains to MQTT
  description: ''
  trigger:
  - platform: state
    entity_id:
    - sensor.total_power
  condition: []
  action:
  - service: mqtt.publish
    data_template:
      payload: |
        {
          "grid": {
            "power": {{ (states('sensor.total_power') | float(0)) }},
            "voltage": {{ (states('sensor.house_volts') | float(0)) }},
            "energy_forward": {{ (states('sensor.total_import_kwh') | float(0)) }},
            "energy_reverse": {{ (states('sensor.total_export_kwh') | float(0)) }}
          }
        }
      topic: homeassistant/energy/grid

mdkeil, 2024-03-17:
pls use <> tags for the corect indent and better readability.

jezzaf, 2024-03-17:
sorry. but as you sent this..

so i restarted everything for the who know how many times and the error is gone 

2024-03-17 07:25:16.809075500 DEBUG:root:Grid: -207.0 W - 234.6 V - -0.9 A
2024-03-17 07:25:21.817318500 DEBUG:root:Grid: -279.4 W - 234.5 V - -1.2 A
2024-03-17 07:25:26.829361500 DEBUG:root:Grid: -195.4 W - 234.6 V - -0.8 A
2024-03-17 07:25:31.834653500 DEBUG:root:Grid: -235.2 W - 234.7 V - -1.0 A

it still doesn't populate the AC input in the dashboard though. any thoughts on this?

[image]

[image]

mdkeil, 2024-03-17:
the default update rate in VRM I think is 15min. Maybe you have to wait a little bit longer?

jezzaf, 2024-03-17:
thank you for all the help. 
still nothing in the dashboard yet.

mr-manuel, 2024-03-17:
From the logs I see, that you did not download the latest script from GitHub else this error would look different:

> ```
> 2024-03-16 23:40:00.179783500 ERROR:root:Exception occurred: 'voltage'
> ```

If you are using an ESS system you have to select the grid meter in the GUI under Settings -> ESS -> Grid metering. Check also Settings ->  System Setup -> AC Input 1.

jezzaf, 2024-03-17:
i defiantly did download it. maybe i didn't do one of the commands correctly. 

don't have and ESS system 

the system settings is set to grid. 

thanks

mr-manuel, 2024-03-17:
Do you have a MultiPlus or Quattro? Why does it say off?

jezzaf, 2024-03-17:
no don't have either. just a bunch of DC equipment. and then hoping to use it to monitor the AC side also. 
the AC loads is populated by a carlo gavazzi. 

i would like to get one but need to work out what is approved in Australia.

mr-manuel, 2024-03-17:
Therefore you don't see anything. You need a MultiPlus or Quattro to display this values properly.

Or you could try if this is working in your case: https://github.com/mr-manuel/venus-os_dbus-multiplus-emulator/issues/2#issuecomment-1966692693

closed 2024-03-17 by mr-manuel

jezzaf, 2024-03-18:
thanks, unfortunately that didn't work, 
it fixed the issue of not showing the load on the AC side but it made the ac load and solar the same reading. 
it also stopped my MPPT charger from charging the batteries. 
thanks for all the help with the rest