mr-manuel/venus-os_dbus-mqtt-grid #19: Alternierende Werte im Akkubetrieb

fellek, 2024-03-29:
Setup
- Easysolar 2 GX (Multiplus 2 GX + MPPT)
- dbus-mqtt-grid
- Homeassistant (MQTT Broker)
- Tasmota IR Reader (Smartmeter)

Die Werte vom Zähler kommen korrekt im im Wechselrichter an.

Aktuell habe ich das Problem: Wenn keine Sonne scheint, soll der Akku genutzt werden, damit Netzbezug auf 0W geregelt wird.

Obwohl die Grundlast konstant bei 300W liegt, wird vom Victronsystem eine alternierende Leistung zwischen -600W bis 600W abgegeben. 
[image]

für mich sieht es aus, als ob die Werte zu träge vom Victronsystem verarbeitet werden.

Kennt jemand das Problem?
Leider fehlt mir aktuell der Ansatz, wo ich nach Fehlern suchen kann (Logs liefern auch im Debug-Mode keine relevanten Infos)

Ich würde gerne ausschließlich, dass es mit dem Adapter zusammenhängt.

mr-manuel, 2024-03-30:
Wie oft verschickst du denn den Wert per MQTT? Laut Screenshot alle 2 Sekunden? Der MQTT Adapter ist nicht das Problem, wenn, dann eine Einstellung im System. In dem Fall ist die Frage in der Victron Community besser aufgehoben.

fellek, 2024-03-30:
Danke für die Rückmeldung, ja die Werte werden verschickt, sobald sie vom Zähler kommen, das ist zwischen 0.5-2 Sekunden.

wenns wohl nicht am Adapter liegt, schaue ich nochmal weiter.

closed 2024-03-30 by mr-manuel