mr-manuel/venus-os_dbus-mqtt-grid #20: Problem with JSON "Minimum required with L1, L2, L3"

buhao75, 2024-04-10:
Hello,
Unfortunately the 3 phases are not shown in my Victron Multiplus GX when using JSON.
Minimum required with L1, L2, L3 does not show any phases but only the main power value:
```
{
    "grid": {
        "power": 0.0,
        "L1": {
            "power": 0.0
        },
        "L2": {
            "power": 0.0
        },
        "L3": {
            "power": 0.0
        }
    }
}
```

Minimum required works as expected:
```
{
    "grid": {
        "power": 0.0
    }
}
```

I would be happy to retrieve a message with some hints.
Thanks and all the best,
Michael

mr-manuel, 2024-04-10:
Did you troubleshoot? What does the log say? How does your payload look like, if you check with MQTT Explorer?

buhao75, 2024-04-10:
Thank you very much for your very fast reply. MQTT Explorer helped me to find the wrong syntax in the MQTT-JSON Format.

closed 2024-04-10 by buhao75