mr-manuel/venus-os_dbus-mqtt-grid #22: Energy Meter is not detected as Grid Meter

oeysteinhansen, 2024-04-29:
I have been successful and using your MQTT Energy Meter script.
But it is not identified as being a Grid Meter. It is instead identified, I think, as a Load Meter.

As shown in the included image. The "AC Loads" box is populated by the MQTT data, not the "AC Input" box.
The system gives me a Warning and is not able to properly work with the Dynamic ESS feature.

Is this a configuration issue. Or is the issue in the MQTT Energy Meter script.

[image]

mr-manuel, 2024-04-29:
Did you look in the closed issues? This is 100% a config issue.

oeysteinhansen, 2024-04-29:
Tank you, I just needed the push in the right direction. 
Saying it was a configuration issue, reading the other relevant closed issues, and having a new look at the configuration.
I needed to set the "System setup" -> "AC input 1" from "Not available" to "Grid". 
This seems to have solved it. I will now run some testing.
[image]

closed 2024-04-29 by oeysteinhansen