mr-manuel/venus-os_dbus-mqtt-grid #25: Problems with MQTT

derGreyson, 2024-08-05:
Moin, 
i have some problems, I use Homeassistant and push/public the Gridinformation to Venus.
My problem is that the Dbus-Mqtt-grid doesnt connect to my mosquito broker.

2024-08-05 23:00:34: New connection from 192.168.178.153:44123 on port 1883.
error: received null username or password for unpwd check

can anybody help me?

mr-manuel, 2024-08-06:
Seems that your MQTT broker is password protected, but you don't configured any password in the config.ini

derGreyson, 2024-08-06:
It is password protected and entered in the config. It is after the
username.

mr-manuel, 2024-08-06:
It is hard to help you without config and driver log. Please post your config (mask the password) and post the driver logs.

derGreyson, 2024-08-06:
Habe nur Nutzername und Passwort und Broker adress geändert.

Config.ini

; default: WARNING
logging = WARNING

; Device name
; default: MQTT Grid
device_name = MQTT Grid

; Device type
; grid = Grid
; genset = Generator
; acload = AC Load (does not impact calculations, shows only under devices)
device_type = grid

; Device VRM instance
; default: 31
device_instance = 31

; Specify after how many seconds the driver should exit (disconnect), if no new MQTT message was received
; default: 60
; value to disable timeout: 0
timeout = 60

; used when no voltage is received
voltage = 230

[MQTT]
; IP addess or FQDN from MQTT server
; "localhost" is allowed, can be used if mqtt server is enabled in venusOS
broker_address = homeassistant.local

; Port of the MQTT server
; default plaintext: 1883
; default TLS port: 8883
broker_port = 1883

; Enables TLS
; 0 = Disabled
; 1 = Enabled
;tls_enabled = 1

; Absolute path to the Certificate Authority certificate file that is to be treated as trusted by this client
;tls_path_to_ca = /data/keys/mosquitto.crt

; Disables verification of the server hostname in the server certificate
; 0 = Disabled
; 1 = Enabled
;tls_insecure = 1

; Username used for connection
;username = mqtt_user

; Password used for connection
;password = pass

; Topic where the grid data as JSON string is published
; minimum required JSON payload: {"grid": { "power": 0.0 } }
topic = enphase/envoy-s/meters

derGreyson, 2024-08-06:
Homeassistant Broker 
Protokoll:

2024-08-06 17:06:46: New connection from 192.168.178.153:33885 on port 1883.
error: received null username or password for unpwd check
2024-08-06 17:06:46: Client MqttGrid_31 disconnected, not authorised.

derGreyson, 2024-08-06:
root@einstein:~# tail -n 100 -F /data/log/dbus-mqtt-grid/current | tai64nlocal
2024-08-06 14:23:13.292034500 *** starting dbus-mqtt-grid ***
2024-08-06 14:24:14.933919500 WARNING:root:Waiting since 60 seconds for receiving first data...
2024-08-06 14:24:14.935014500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2024-08-06 14:24:15.106956500 *** starting dbus-mqtt-grid ***
2024-08-06 14:25:16.761732500 WARNING:root:Waiting since 60 seconds for receiving first data...
2024-08-06 14:25:16.761742500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2024-08-06 14:25:17.135228500 *** starting dbus-mqtt-grid ***

4

derGreyson, 2024-08-06:
2024-08-06 15:09:43.310862500 *** starting dbus-mqtt-grid ***
2024-08-06 15:09:44.996884500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:09:44.996893500
2024-08-06 15:09:44.998301500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:09:44.999253500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:09:44.999927500 WARNING:root:MQTT client: Trying to reconnect
2024-08-06 15:09:46.028003500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:09:46.028013500
2024-08-06 15:09:46.029690500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:09:46.030376500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:09:48.045766500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:09:48.045774500
2024-08-06 15:09:48.047231500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:09:48.048033500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:09:52.063332500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:09:52.063342500
2024-08-06 15:09:52.064653500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:09:52.065314500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:00.080428500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:00.080438500
2024-08-06 15:10:00.081657500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:00.082370500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:02.469889500 *** starting dbus-mqtt-grid ***
2024-08-06 15:10:04.274179500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:04.274189500
2024-08-06 15:10:04.275883500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:04.281084500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:04.281094500 WARNING:root:MQTT client: Trying to reconnect
2024-08-06 15:10:05.303615500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:05.303624500
2024-08-06 15:10:05.304852500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:05.305513500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:07.322128500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:07.322137500
2024-08-06 15:10:07.323815500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:07.324793500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:11.341314500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:11.341323500
2024-08-06 15:10:11.342583500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:11.343259500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:19.360125500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:19.360134500
2024-08-06 15:10:19.361548500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:19.362256500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:24.230110500 *** starting dbus-mqtt-grid ***
2024-08-06 15:10:25.810665500 INFO:root:MQTT client: Connecting to broker homeassistant.local on port 1883
2024-08-06 15:10:25.852940500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:10:25.855311500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:25.855321500
2024-08-06 15:10:25.856705500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:25.857371500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:25.858026500 WARNING:root:MQTT client: Trying to reconnect
2024-08-06 15:10:26.897056500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:26.897065500
2024-08-06 15:10:26.898179500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:26.898966500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:28.916023500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:28.916032500
2024-08-06 15:10:28.917135500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:28.917782500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:30.859122500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:10:32.939698500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:32.939708500
2024-08-06 15:10:32.940935500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:32.941572500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:35.865093500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:10:40.870584500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:10:40.957136500 ERROR:root:MQTT client: Failed to connect, return code 5
2024-08-06 15:10:40.957144500
2024-08-06 15:10:40.958728500 WARNING:root:MQTT client: Got disconnected
2024-08-06 15:10:40.959755500 WARNING:root:MQTT client: Unexpected MQTT disconnection. Will auto-reconnect
2024-08-06 15:10:45.876844500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:10:50.880762500 INFO:root:Waiting 5 seconds for receiving first data...
[... cut ...]
2024-08-06 15:11:10.901359500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:11:15.905908500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:11:20.911667500 INFO:root:Waiting 5 seconds for receiving first data...
2024-08-06 15:11:25.914575500 WARNING:root:Waiting since 60 seconds for receiving first data...
2024-08-06 15:11:25.915354500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2024-08-06 15:11:26.088620500 *** starting dbus-mqtt-grid ***

derGreyson, 2024-08-06:
I have 2 Users who work but both not with this config..

is there any maximum lenght of allowed passwords?

[17:17:12] INFO: Successfully send discovery information to Home Assistant.
2024-08-06 17:17:13: New connection from 192.168.178.153:47849 on port 1883.
error: received null username or password for unpwd check
2024-08-06 17:17:13: Client MqttGrid_31 disconnected, not authorised.
2024-08-06 17:17:16: New connection from 172.30.32.1:47535 on port 1883.
2024-08-06 17:17:16: New client connected from 172.30.32.1:47535 as 2wxhBMna9zJdvEY1Bf9T7V (p2, c1, k60, u'mqtt_user').
2024-08-06 17:17:21: New connection from 192.168.178.153:33257 on port 1883.
error: received null username or password for unpwd check
2024-08-06 17:17:21: Client MqttGrid_31 disconnected, not authorised.
[17:17:22] INFO: Successfully send service information to the Supervisor.
2024-08-06 17:17:29: New connection from 192.168.178.11:61662 on port 1883.
2024-08-06 17:17:29: New client connected from 192.168.178.11:61662 as DVES_5BF49A (p2, c1, k30, u'homeassistant').
2024-08-06 17:17:37: New connection from 192.168.178.153:57165 on port 1883.

derGreyson, 2024-08-06:
My mistake was **;**tls_insecure = 1 has to be deleted

mr-manuel, 2024-08-06:
All solved?

derGreyson, 2024-08-07:
Yeah thx i was stupid.

derGreyson, 2024-08-07:
Last question.

my Hitchi has 10 seconds delay cause telemetry. Is there an option to get it faster?

closed 2024-08-07 by derGreyson