mr-manuel/venus-os_dbus-mqtt-grid #36: Enhancement to differentiate between Grid & AC Loads

pellenbeck, 2025-01-30:
Would it be possible to differentiate between a Grid Meter and an AC Loads Meter by configuration?
In Venus OS, a distinction can be made between the following cases:

Grid meter: Measures grid power flow
AC loads: Measures power consumed by AC loads
PV inverter: Measures solar power production
Generator: Measures generator output5

mr-manuel, 2025-01-30:
Either I do not get your question or you did not check the config file.

https://github.com/mr-manuel/venus-os_dbus-mqtt-grid/blob/351e5cd37dff477d7de1c69e0c9e7bb25c931ca8/dbus-mqtt-grid/config.sample.ini#L18-L22

pellenbeck, 2025-01-30:
> Either I do not get your question or you did not check the config file.

You're absolutely right, or rather, you didn't get it wrong. I didn't check the config at that point.
It is not possible to adjust the assignment in the Venus OS frontend; but that is not a fault of your script.

closed 2025-01-30 by pellenbeck