mr-manuel/venus-os_dbus-mqtt-grid #37: Need an option per mqtt to stop and remove the GRID-Device in VenusOS

pos-ei-don, 2025-02-19:
Need an option per mqtt to stop and remove the GRID-Device in VenusOS.

I know that I can disable sending mqtt messages and just wait for a timeout,
but i need it to remove quickly as i use this as a failover if the main grid meter is 
not available (during firmware updates, etc...)

as it produces high currents sometimes by telling ess wrong values, i need it to disappear more quickly.

I tried to just send it an empty {} json, but this did not do the trick.
Is there a better way?

mr-manuel, 2025-02-20:
You could try to set the timeout to 1 or 2 seconds. Or one second more then your send cycle is. Let me know if that works for you.

pos-ei-don, 2025-02-20:
Yes, works ok. Sometimes because of network lag, I get an Update too slow so the device junos from connected to not connected, but at the end it works.

Thank you for the idea!!

closed 2025-02-20 by pos-ei-don

mr-manuel, 2025-02-20:
I think even if you set a 6 second timeout it should be ok. Normally, if the data is delayed more then 5 seconds, then you already have a problem which makes the system unstable. In this case better to disconnect the grid meter in your case.