mr-manuel/venus-os_dbus-mqtt-grid #39: Misunderstand of energy_(forward/reverse) field

MoMoPouetDev, 2025-03-25:
Hi,
1) - I think i don't understand the energy field.

Is it the total/lifetime energy that I bought and sold?

Because here is my info :

[image]

Is it normal behavior?

2) - Just for my information, when I send through MQTT the energy that I sell, I send with minus sign 

[image]

[image]

Is it ok like that?

Thanks,
Morgan

mr-manuel, 2025-03-25:
I updated the readme.

```json
{
    "grid": {
        "power": 0.0,                <-- watts
        "voltage": 0.0,              <-- volts
        "current": 0.0,              <-- amps
        "energy_forward": 0.0,       <-- imported/bought energy (lifetime), positive value in kWh
        "energy_reverse": 0.0,       <-- exported/sold energy (lifetime), positive value in kWh
        "L1": {
            "power": 0.0,            <-- watts
            "voltage": 0.0,          <-- volts
            "current": 0.0,          <-- amps
            "frequency": 0.0000,     <-- Hz
            "power_factor": 0.0,
            "energy_forward": 0.0,   <-- imported/bought energy (lifetime), positive value in kWh
            "energy_reverse": 0.0    <-- exported/sold energy (lifetime), positive value in kWh
        },
        "L2": {
            "power": 0.0,            <-- watts
            "voltage": 0.0,          <-- volts
            "current": 0.0,          <-- amps
            "frequency": 0.0000,     <-- Hz
            "power_factor": 0.0,
            "energy_forward": 0.0,   <-- imported/bought energy (lifetime), positive value in kWh
            "energy_reverse": 0.0    <-- exported/sold energy (lifetime), positive value in kWh
        },
        "L3": {
            "power": 0.0,            <-- watts
            "voltage": 0.0,          <-- volts
            "current": 0.0,          <-- amps
            "frequency": 0.0000,     <-- Hz
            "power_factor": 0.0,
            "energy_forward": 0.0,   <-- imported/bought energy (lifetime), positive value in kWh
            "energy_reverse": 0.0    <-- exported/sold energy (lifetime), positive value in kWh
        }
    }
}
```

MoMoPouetDev, 2025-03-25:
Ok so that's what I did.
The informations in my pictures seems ok then, what do you think?

mr-manuel, 2025-03-25:
Yes, if you have a big system that prodced 29 GWh in it's lifetime.

MoMoPouetDev, 2025-03-25:
Sorry this is not what I'm saying.

What I send
Energy_forward : 29152110
Energy_reverse : 676765

It is the energy I've been buying since I bought my house, is that correct?
So yes in energy L1 this is correct if it's correct.

But value on ac total seems cut.

And total energy reverse does not appears?

Thanks,
Again this is just to understand your work.

mr-manuel, 2025-03-25:
All correct. Only the display of the value is cut, since the filed is to short. Change to GUIv2 and you see all.

Energy reverse is not displayed by Victron, maybe in a future release of them.

MoMoPouetDev, 2025-03-25:
Actually, I don't think this is a normal value on consumption.

[image]

Not consume 111Kwh in 1hr.
I send every sec mqtt value, is it too much?

MoMoPouetDev, 2025-03-25:
Or maybe it's because of the total energy send in first time

mr-manuel, 2025-03-25:
Very likely you provide the energy values in watts instead of kilowatts.

MoMoPouetDev, 2025-03-25:
Ok will check 
Thanks

mr-manuel, 2025-03-26:
No, ask Google or the Victron Community.

MoMoPouetDev, 2025-03-26:
Yes I think this is ok know 
Thanks for your help