mr-manuel/venus-os_dbus-mqtt-grid #40: Typo capitalization - ac vs. Ac

olifrieda, 2025-03-26:
There are some typos in `dbus-mqtt-grid.py`: should be first upper case

https://github.com/mr-manuel/venus-os_dbus-mqtt-grid/blob/668d174f0ab7d9c6a5f1aed8d6d83bb9231ae03a/dbus-mqtt-grid/dbus-mqtt-grid.py#L379

```
                    self._dbusservice["/ac/L1/PowerFactor"] = round(grid_L1_power_factor, 3) if grid_L1_power_factor is not None else None
                                       ^^^^
```

https://github.com/mr-manuel/venus-os_dbus-mqtt-grid/blob/668d174f0ab7d9c6a5f1aed8d6d83bb9231ae03a/dbus-mqtt-grid/dbus-mqtt-grid.py#L395

```
                    self._dbusservice["/ac/L2/PowerFactor"] = round(grid_L2_power_factor, 3) if grid_L2_power_factor is not None else None
                                       ^^^^
```

https://github.com/mr-manuel/venus-os_dbus-mqtt-grid/blob/668d174f0ab7d9c6a5f1aed8d6d83bb9231ae03a/dbus-mqtt-grid/dbus-mqtt-grid.py#L404
```
                    self._dbusservice["/ac/L3/PowerFactor"] = round(grid_L3_power_factor, 3) if grid_L3_power_factor is not None else None
                                       ^^^^
```

olifrieda, 2025-03-31:
Maybe you should add a notice in the readme, that this option is only allowed to be set on a 3-phase-system. Otherwise venus os will throw an error.

mr-manuel, 2025-03-31:
Which option and what error?

mr-manuel, 2025-04-27:
@olifrieda I was not able to reproduce it with your payload. I changed the code a bit. Could you retry with the latest nightly/file in the repo?