mr-manuel/venus-os_dbus-mqtt-grid #44: configparser.py isnt parsing MQTT username and password correctly

SirTekkit, 2025-05-22:
Both stable and nightly releases are showing the same error. The service is not crashing at least. But it is retrying to parse the config every 1~2sec.

```
2025-05-22 19:54:32.683661500 *** starting dbus-mqtt-grid ***
2025-05-22 19:54:34.204356500 Traceback (most recent call last):
2025-05-22 19:54:34.204362500   File "/data/etc/dbus-mqtt-grid/dbus-mqtt-grid.py", line 591, in <module>
2025-05-22 19:54:34.204367500     main()
2025-05-22 19:54:34.204368500   File "/data/etc/dbus-mqtt-grid/dbus-mqtt-grid.py", line 486, in main
2025-05-22 19:54:34.204372500     if "username" in config["MQTT"] and "password" in config["MQTT"] and config["MQTT"]["username"] != "" and config["MQTT"]["password"] != "":
2025-05-22 19:54:34.204380500   File "/usr/lib/python3.8/configparser.py", line 1255, in __getitem__
2025-05-22 19:54:34.204485500     return self._parser.get(self._name, key)
2025-05-22 19:54:34.204488500   File "/usr/lib/python3.8/configparser.py", line 799, in get
2025-05-22 19:54:34.204492500     return self._interpolation.before_get(self, section, option, value,
2025-05-22 19:54:34.204496500   File "/usr/lib/python3.8/configparser.py", line 395, in before_get
2025-05-22 19:54:34.204500500     self._interpolate_some(parser, option, L, value, section, defaults, 1)
2025-05-22 19:54:34.204528500   File "/usr/lib/python3.8/configparser.py", line 442, in _interpolate_some
2025-05-22 19:54:34.204533500     raise InterpolationSyntaxError(
2025-05-22 19:54:34.204535500 configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: '%'
```

For reference my `config.ini` contains this for the username:
```
; Username used for connection
username = venusOS
```

mr-manuel, 2025-05-22:
Do you have a % in your password? If yes try with %% for %.

SirTekkit, 2025-05-23:
There was indeed a % included in the password. Trying %% does fix it the issue.
I should've thought of that myself but thank you so much. Im probably still gonna change that % to something else.

MQTT Grid is running now and displaying data in the venus GUI! after fixing commas in my JSON payload :)

closed 2025-05-23 by SirTekkit