mr-manuel/venus-os_dbus-mqtt-grid #48: Issue with "retained" MQTT messages

makkie2002, 2025-09-03:
I have been using your code for several months now without issues ... until today :) :(.

This is the issue:
- I am feeding in the grid power data (from the Belgian P1 port) through MQTT.
- I have the "retain" flag on for these messages (for some now-unknown reason).
- If now my grid-meter is not updating the MQTT messages anymore, the script restarts after the 60s timeout, it subscribes to the MQTT topic, it receives the (old) retained value and sends it to Victron  ...
- In my case the Victron got fully confused and was discharging the battery at full power ....

So maybe to solve this issue we could check if after restarting the service the first received value is ignored, so that we are sure that we are not dealing with retained messages ...

Thanks!

mr-manuel, 2025-09-04:
Solving the issue on the root would be the better choice. Do not retain data for values that are not valid anymore after a short amount of time.

makkie2002, 2025-09-04:
> Solving the issue on the root would be the better choice. Do not retain data for values that are not valid anymore after a short amount of time.

Hmmm, yes and no. I had to do some "advanced" bug finding to find out what the real issue was. Luckily I have the skills, but not all thinkerers have these skills. Just switching of the "retain" flag when sending messages was not enough, because the MQTT server was server the last kept message with the retain flag on. I actually had to send an empty message (with retain flag on) for the MQTT server to forget about this. So I can imagine other poeple running into the same issue as I did. And I think by capturing this "edge case" you could improve on the overall robustness of the service 😃 ... Just have the script forget about the first received power message should do the trick.